zmkfirmware/zmk #3051: Print firmware and configuration information

a-b, 2025-08-31:
It would be helpful to have a unique key combination that allows printing of the uploaded firmware information and configuration details. This way, users can press the keys `Magic + V` and view information about the uploaded firmware and settings in a text editor.

a-b, 2025-09-09:
Thank you for clarifying this question. I am considering the simplest option, which is to simulate a key press. This way, anyone could press this magic combination and see the version and perhaps other essential config details in their favorite editor.

nmunnich, 2025-09-09:
Realistically, it makes the most sense to me if this information was simply exposed to ZMK Studio and can be displayed there. I find it doubtful that users would actually place such a key in their keymap.

mhantsch, 2025-09-26:
> Thank you for clarifying this question. I am considering the simplest option, which is to simulate a key press. This way, anyone could press this magic combination and see the version and perhaps other essential config details in their favorite editor.

This will only work if the keymap on the host OS is set to US. For users with other host keymaps it would only show garbled text.